jrsteensen/OpenHornet #552: Create Drawing: OH5A7A1 - ASSY, SNSR PANEL

jrsteensen, 2022-09-14:
Create drawing OH5A7A1 - ASSY, SNSR PANEL.

Lazarus94, 2023-04-26:
IN PROGRESS

Lazarus94, 2023-04-26:
IN CHECK

[OH5A7A1-1 - ASSY, SNSR PANEL v1.pdf](https://github.com/jrsteensen/OpenHornet/files/11331703/OH5A7A1-1.-.ASSY.SNSR.PANEL.v1.pdf)

jrsteensen, 2023-04-27:
IN BACKCHECK

[image]

Lazarus94, 2023-04-27:
IN BACKCHECK

The switch bat is both FN10 and 24?

[OH5A7A1-1 - ASSY, SNSR PANEL v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11343250/OH5A7A1-1.-.ASSY.SNSR.PANEL.v2.pdf)

jrsteensen, 2023-05-03:
No idea what is happening there but....

[image]

Lazarus94, 2023-05-07:
IN BACKCHECK

[OH5A7A1-1 - ASSY, SNSR PANEL v3.pdf](https://github.com/jrsteensen/OpenHornet/files/11414090/OH5A7A1-1.-.ASSY.SNSR.PANEL.v3.pdf)

jrsteensen, 2023-05-07:
Added to RC Release Package and closed.

closed 2023-05-07 by jrsteensen