jrsteensen/OpenHornet #554: Create Drawing: OH5A9A1 - ASSY, KY58 PANEL

jrsteensen, 2022-09-14:
Create drawing OH5A9A1 - ASSY, KY58 PANEL.

Lazarus94, 2023-04-26:
IN CHECK

[OH5A9A1-1 - ASSY, KY58 PANEL v1.pdf](https://github.com/jrsteensen/OpenHornet/files/11331217/OH5A9A1-1.-.ASSY.KY58.PANEL.v1.pdf)

Lazarus94, 2023-04-26:
IN CHECK

Slightly Updated

[OH5A9A1-1 - ASSY, KY58 PANEL v3.pdf](https://github.com/jrsteensen/OpenHornet/files/11331579/OH5A9A1-1.-.ASSY.KY58.PANEL.v3.pdf)

kbastronomics, 2023-04-26:
see boxed area.
it looks like it's projecting the pins of the plug assembly, shouldn't it be a single projection line?  
Might be a fusion limit just pointing it out

[image]

Lazarus94, 2023-04-26:
Maybe worth adding a note about mounting of the battery cover on the panel?

Lazarus94, 2023-04-26:
@kbastronomics #456

kbastronomics, 2023-04-26:
Thank you. my bad.

jrsteensen, 2023-04-27:
IN BACKCHECK

[image]

Lazarus94, 2023-04-27:
IN BACKCHECK

[OH5A9A1-1 - ASSY, KY58 PANEL v5.pdf](https://github.com/jrsteensen/OpenHornet/files/11343368/OH5A9A1-1.-.ASSY.KY58.PANEL.v5.pdf)

jrsteensen, 2023-05-03:
Added to RC Release Package and Closed

closed 2023-05-03 by jrsteensen