jrsteensen/OpenHornet #555: Create Drawing: OH5A10A1 - ASSY, RIGHT CIRCUIT BREAKERS

jrsteensen, 2022-09-14:
Create drawing OH5A10A1 - ASSY, RIGHT CIRCUIT BREAKERS.

Lazarus94, 2023-04-25:
IN PROGRESS

Lazarus94, 2023-04-26:
IN CHECK 

[OH5A10A1-1 - ASSY, RIGHT CIRCUIT BREAKERS v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11330621/OH5A10A1-1.-.ASSY.RIGHT.CIRCUIT.BREAKERS.v2.pdf)

Lazarus94, 2023-04-26:
IN CHECK

Slightly Updated

[OH5A10A1-1 - ASSY, RIGHT CIRCUIT BREAKERS v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11331568/OH5A10A1-1.-.ASSY.RIGHT.CIRCUIT.BREAKERS.v4.pdf)

jrsteensen, 2023-04-27:
IN BACKCHECK

[image]
[image]

Lazarus94, 2023-04-27:
IN BACKCHECK

[OH5A10A1-1 - ASSY, RIGHT CIRCUIT BREAKERS v5.pdf](https://github.com/jrsteensen/OpenHornet/files/11343409/OH5A10A1-1.-.ASSY.RIGHT.CIRCUIT.BREAKERS.v5.pdf)

jrsteensen, 2023-05-03:
Added to RC Release Package and Closed.

closed 2023-05-03 by jrsteensen