jrsteensen/OpenHornet #556: Create Drawing: OH5A10A2 - ASSY, CANOPY SWITCH

jrsteensen, 2022-09-14:
Create drawing OH5A10A2 - ASSY, CANOPY SWITCH.

Lazarus94, 2023-04-01:
IN PROGRESS

Lazarus94, 2023-04-16:
IN CHECK

[OH5A10A2-1 - ASSY, CANOPY SWITCH  v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11242891/OH5A10A2-1.-.ASSY.CANOPY.SWITCH.v2.pdf)

jrsteensen, 2023-04-16:
IN BACKCHECK

[image]
[image]

Lazarus94, 2023-04-16:
IN BACKCHECK

[OH5A10A2-1 - ASSY, CANOPY SWITCH v5.pdf](https://github.com/jrsteensen/OpenHornet/files/11243025/OH5A10A2-1.-.ASSY.CANOPY.SWITCH.v5.pdf)

jrsteensen, 2023-04-16:
Added to RC Release Package and closed.

closed 2023-04-16 by jrsteensen