jrsteensen/OpenHornet #566: Create Drawing: OHC0005 - ASSY, MAG HELD SWITCH

jrsteensen, 2022-09-21:


jrsteensen, 2022-09-24:
[OHC0005 - ASSY, MAG HELD SWITCH_-_v2.pdf](https://github.com/jrsteensen/OpenHornet/files/9639087/OHC0005.-.ASSY.MAG.HELD.SWITCH_-_v2.pdf)

In check

jrsteensen, 2022-09-24:
Updated with switch modification procedure.

[OHC0005 - ASSY, MAG HELD SWITCH_-_v4.pdf](https://github.com/jrsteensen/OpenHornet/files/9639187/OHC0005.-.ASSY.MAG.HELD.SWITCH_-_v4.pdf)

russellmm, 2022-09-25:
Page 1 shows all switches as "on" in the middle postion under the description. The folliing pages show those positions as "off". Might confuse some but unlikely to affect anything.

jrsteensen, 2022-09-25:
Fixed in drawing v5.

closed 2022-09-25 by jrsteensen

jrsteensen, 2022-09-25:
Added bond note for standoff in v6

[image]