jrsteensen/OpenHornet #580: Standby Alt, Encoder mount change required

Ben-F111, 2022-10-24:
### Summary:
On test asem of the standby Alt it is noted there is not enough thread to secure the top nut, aprox 1mm of matrial is required to be removed from the underside of the Encoder mount 

### Expected Results/How It Should Work:
there should be at least enough thread to enter the nut

### Actual Results/How It Does Work:
there is no thread above the plastice to start turning the nut

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (F360)

### Type:
*Check one item.*
- [X] Bug

### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [X] Lower Instrument Panel (LIP) - Standby Gauges

### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

Lazarus94, 2023-03-26:
Fixed in [OH2A7A1-200 - ALTIMETER] V8

jrsteensen, 2023-03-26:
Review Comments

- [ ] Need to update up through the console level - should be fixed in LIP vXXXX with latest components.

Lazarus94, 2023-03-26:
> ### Review Comments
> * [ ]  Need to update up through the console level - should be fixed in LIP vXXXX with latest components.

Will do.

jrsteensen, 2023-03-26:
Fixed in LIP v96

closed 2023-03-26 by jrsteensen