jrsteensen/OpenHornet #581: Standby Alt (Pressure window) - needs moving

Ben-F111, 2022-10-24:
### Summary:
it was noted during test asem of the Standby ALT, that the window allowed for the Pressure TEXT on the lower portion of the Standby ALT does not align with the text position printed on the OLED

### Expected Results/How It Should Work:
The OLED text needs to be positioned in the center of the face plate window, the window needs to move up 1 mm to alight with the text, this is the lowest available position in the usable screen area of the OLED

Also if possible due to parallax the face plate needs to be as thin as it can be suggest investigating the thinnest Romark, or Resin printed and remove all not required depth

### Actual Results/How It Does Work:
Due to the available "print" area of the OLED the actual text is slightly higher than the cut out

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (F360)
### Type:
*Check one item.*
- [X] Bug

### Applicable End Item:
*Check one item.*
- [X] Lower Instrument Panel (LIP) - Standby ALT

### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

Lazarus94, 2023-03-26:
Fixed in [OH2A7A1-200-ALTIMETER] V9. Moved cutout up by 1mm. Note: investigate making faceplate thinner.

jrsteensen, 2023-03-26:
### Review Comments

- [ ] Need to update up through the console level - should be fixed in LIP vXXXX with latest components.

jrsteensen, 2023-03-26:
Fixed in LIP v96

closed 2023-03-26 by jrsteensen