jrsteensen/OpenHornet #582: Standby Gauged Backlight LED Screen Print

Ben-F111, 2022-10-27:
### Summary:
Investigation required to confirm the orientation of the FLAT EDGE for the WS2812B LEDs used for the back lighting

### Expected Results/How It Should Work:
Screen print and LED pin out should match the Physical LED Pin out, the "Flat edge" shown on the screen print should match the flat edge of the LED

### Actual Results/How It Does Work:
During testing and install of WS2812B LEDs it has been found this screen print may be 180deg out, de soldering and revering the LED's to the screen print results in corrct operation of the test sets

investigation required per the below

1 are there more than one brand type of WS2821B LEDs with differing Pin outs
2 if not, investigate and confirm the current screen print is OK or needs fllipping
3 if yes, uncover a way of marking thew PCB to allow for both

current investigation, including NEOPixel and Adifuit NEOPixel SHow the DATA OUT Pin on the FLAT EDGE, the current OH design has this on the DATA IN Wire.

Current 
P4 DATA IN Flat Edge
P3 VDD- 5VDC
P2 GND
P1 DATA OUT
(Does not work when installed flat edge to screen print)

Should be
P4 DATA OUT - Flat Edge
P3 GND 
P2 VDD- 5VDC
P1 DATA IN
[image]

[image]

Uploading IMG_2691.MOV…

[image]

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] ECAD (PCB Design or other electrical hardware)

### Type:
*Check one item.*
- [X] Bug
### Applicable End Item:
*Check one item.*
      
- [X] Lower Instrument Panel (LIP) - Standby Gauges 

### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

Ben-F111, 2022-10-27:
https://user-images.githubusercontent.com/67782885/198403699-17397b02-1e30-4911-b6e6-85c4d2f5fbcb.MOV

Exprezz0, 2023-01-31:
@Ben-F111 These are the WS2812B, seen on the backlight boards.

[image]

But can confirm, Pin #2 is suppose to be Ground.  
WP154A4, https://www.kingbright.com/attachments/file/psearch/000/00/00/L-154A4SUREQBFZGEW(Ver.11A).pdf

The footprint and silkscreen are correct; the pins were just connected to the wrong nets.  I'll fix it and get it pushed up after Issue #592 gets reviewed.

Exprezz0, 2023-04-22:
Fixed with commit: 2de58f52bfa469f13f1cb1c4f65ecbc64b1d1c8a.  Will be implemented with #592.  The 2x 4-pin LEDs are based on this: https://www.adafruit.com/product/1938.  

Standby Gauge needs to be transitioned and updated to K7.  I'll double check BOM and files again, but this issue specifically has been fixed.

closed 2023-04-22 by Exprezz0