jrsteensen/OpenHornet #584: D1 Diode silkscreen incorrect in Hall Effect Sensor board

geeman-nz, 2022-11-10:
### Summary:
The silkscreen for the Hall effect sensor PCB is incorrect. It displays that the diode should be placed in the incorrect orientation, leading to the polarity being backwards.

### Expected Results/How It Should Work:
The + symbol should be placed on the opposite side to where it is currently placed

### Screenshots:
Note the diode is placed in opposite polarity to the marked silkscreen.
This image was produced by a JLCPCB engineer when this issue was questioned.

[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [X] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [X] Throttle
- [ ] General Software
### Associated Filename(s):
ABSIS_Hall_Sensor

Exprezz0, 2023-01-31:
@geeman-nz Is there any reason why the Diode can't be rotated 180 degrees instead of adjusting the silkscreen?  The + symbol indicates the correct pad.

[image]

Arribe, 2023-03-29:
From ordering experience this is confusing.  Shouldn't the silkscreen reflect the direction of the diode and not which pad is '+'? There are instances for using reversed polarity diodes. 
[image]

Exprezz0, 2023-03-29:
[image]

Will be closed with #592

jrsteensen, 2023-03-30:
@Exprezz0 - for brevity, just add the K to the left side, no arrow. Stays more consistent with single character markings of other polarities. Drop the A. (If you don't mind - if you already implemented it, don't worry about it. Nothing I'm going to lose sleep over.)

Exprezz0, 2023-03-30:
Too easy.  Just figured I'd alleviate all doubt.  Fixed and updated.

closed 2023-04-16 by jrsteensen