jrsteensen/OpenHornet #585: Brass Insert Clearance Holes in Park Brake Mech Body too big

Stang51, 2022-11-17:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
(The clearance holes for the Brass Inserts (93365A112) in the Park Brake Mechanism Body are too large.  The Insert slips right in with next to no pressure.  The result is they will pull out very easy.

### Expected Results/How It Should Work:
(The correct clearance hole for the Brass Insert is   ".1065"    The Current Hole is   ".1406  ")

### Actual Results/How It Does Work:
(These clearance holes for the 93365A112 Brass Inserts are located in the Parking Brake Mechanism Body.  There are 7 Holes that are affected)

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** [image].*

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ X] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [X ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ X] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Assembly - Parking Brake Assembly - OH4A3A2-1  -  ASSY, PARK BRAKE HANDLE v9*
*Assembly Part - OH4A3A2-10  -  PARKING BRAKE MECHANISM BODY:1*

jrsteensen, 2023-03-25:
Fixed in LC v131

closed 2023-03-25 by jrsteensen

Stang51, 2023-03-26:
Good stuff.

How’s the Back alighted PCBs coming along for the Legend panels?

Tejay