jrsteensen/OpenHornet #587: ABSIS_HID_Bus_Master: SOIC-8 wrong angle

Higgins-OH, 2022-11-19:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
After uploading the necessary files on JLCPCB I noticed that the SOIC-8 is rotated 270 degress. As a result the pins don't match the PCB. The rotation should be 180 or 0 degress, depending on the circuits.

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [x] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [x] Upper Instrument Panel (UIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
ABSIS003 ABSIS_HID_BUS_MASTER

Higgins-OH, 2022-11-19:
The "align" command seems to fix the problem. Manfuacturing will tell...

jrsteensen, 2022-11-19:
> The "align" command seems to fix the problem. Manfuacturing will tell...

Align command on what? Sorry, need more detail here.

Higgins-OH, 2022-11-19:
The SMD parts are always far from the PCB in JLCPCB's preview window. I just accidently clicked on the "align" button and every part was at its intended place. Including the SOIC. 
[image]

Exprezz0, 2023-01-31:
This is an issue with JLC, not the BOM/CPL.  Fixed with #592

closed 2023-04-16 by jrsteensen