jrsteensen/OpenHornet #588: ABSIS_MEGA: top pos incorrect?

Higgins-OH, 2022-11-20:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
Manufacturing by JLCPCB not possible due to incorrect part placement. Even the website's "align" button is no help here this time.

### Expected Results/How It Should Work:
The parts need to be placed correctly. Incorrect top pos file?

### Actual Results/How It Does Work:
(Explain how it actually works here. Be very specific.)

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [x] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [x] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
ABSIS_Mega-top-pos.csv (file date: 2021-02-02)

Exprezz0, 2023-01-30:
This has been "fixed" with issue #592.  It's been corrected in the BOM and CPL files and matches the PCB model.  However, when you upload it to JLCPCB.com, it still renders incorrectly.  The problem is on their website and how they analyze their data.  Unfortunately, there's no way of anticipating what components are broken and which ones work.  We also can't anticipate when they decide to fix it.  

We can manually correct the CPL files to rotate so that everything lines up.  This has happened previously, where OH took the steps necessary to make adjustments within the manufacturing files to facilitate the errors on JLCPCB.com.  Then JLC fixed the issues on their website, which left all of the manufacturing files with problems.  Additionally, without any concrete evidence from JLC on how they align their components when the customer uses the web interface, I don't feel comfortable making changes in the CPL and "guessing" the orintation.

I've attached a few images to further explain the situation.  In KiCad, the white dot on the model represents pin 1, which is not the ground.  The ground is pin 5 (top left corner of the model depicted below)

[image]
[image]
[image]

According to JLCPCB.com:
[image]

and I've tested it on their site:
[image]

If the purple dot represents Pin #1, then I would need to rotate it 90 degrees to the **RIGHT**
If the purple dot represents GND (Pin #5), then it would need to be rotated 90 degress to the left.

Adding the dots to the silkscreen would be the answer.  In the meantime, letting their engineers resolve it during the assembly process is probably the safest way to go in you're unsure.

kbastronomics, 2023-01-31:
Please keep in mind,  the JCLPCB's placement information is for reference only,   do not use it for part placement.
They do a DFM analysis and you're suppose to review that.   Changes to the reference page are not captured to my knowledge in the final placement.  

for reference.
[image]

Exprezz0, 2023-01-31:
Correct.  It's highly recommended that everyone selects the "Check final placement" option when placing an order.  But it does help identify some potential issues that could reside within our files.  With this issue specifically though, it belongs to JLC, not Open Hornet.

kbastronomics, 2023-01-31:
True,   I did check by placing an order (then cancelling it) and can confirm both the preview and DRM report show bad placement of U1,  rotated 180 degree's when it came back from DRM check

Exprezz0, 2023-04-22:
Fixed with commit:  111972f86e6a11e4c27b4371f98b5cc966385848.  Will be implemented with #592

closed 2023-04-22 by Exprezz0