jrsteensen/OpenHornet #595: UFC Incorrect Pin Spacing on UFCOptionDigit

kbastronomics, 2022-11-29:
### Summary:
The UFCOptionDigit displays from Tekcreation do not fit on the V5 UFC Board.   Pins are 0.055 in (1.3mm) closer together than the should be.  correct distance is 0.500 in

### Expected Results/How It Should Work:
The UFCOptionDigit should fit the pad precisely.

### Actual Results/How It Does Work:
The UFCOptionDigit does not correctly fit the pads on the PCB due the pads are 0.055" to close,  this causes bending the UFCOptionDigit leg beyond what can be comfortably used

### Screenshots:

*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.

**V5 PCB as manufactured from JCLPCB**
[image]

**Image showing how the UFCOptionDigit fits on the pcb** 
[image]

**Image show sample adjustment to one pin of the footprint**
[image]

**Images shows UFCOptionDigit fitting correctly on a printed copy of the PCB design after fixing the footprint**
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [X] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [X] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*
ufc_v5_main.pro

jrsteensen, 2023-03-25:
This will need to be factored in with the UFC redesign.

jrsteensen, 2023-04-04:
Fixed footprint, will be reflected in UFC v5.5 PCB.

closed 2023-04-04 by jrsteensen