jrsteensen/OpenHornet #596: UFC UFCOptionDigit will not display correctly due to use of GRID sharing of A/B & E/F

kbastronomics, 2022-11-29:
### Summary:
The UFCOptionDigit shares the A/B Grid and E/F Grid to free up space for the DP then leaves a GRID line unused.  because of the some digits will not display correctly.  2,3,5,6,7,8,9,  any digit were the A/B or E/F segements would light. 

### Expected Results/How It Should Work:
digit should display correctly as seen on a UFC in DCS

### Actual Results/How It Does Work:
This is analysis of the schematic,  I can't show this due to the  UFCOptionDigit will not install per my other issue.
I would suggest wiring Segments A-P to Grid 1-16,   wire the C5/D5 ping direct to the S2-Mini

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.

**Digits should display like this**
[image]

This is what will happen with the current way the  UFCOptionDigit is wired
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [X] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [X] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
ufc_v5_main.pro

jrsteensen, 2023-03-25:
UFC Redesign issue.

jrsteensen, 2023-03-28:
Leaving assigned to beta1 for now, but may remove for #644 for after Beta1.

jrsteensen, 2023-04-05:
Closing this as not planned after further discussion with @kbastronomics.

closed 2023-04-05 by jrsteensen