jrsteensen/OpenHornet #598: OBOGS Knob/Rotate Switch - Misaligned

Exprezz0, 2022-11-30:
### Summary:
Alignment of OBOGS Knob on rotary switch **OR** rotation of the Rotary Switch that controls oxygen flow inhibits the direction that the OBOGS Knob arrow points at the "OXY FLOW _ON_ and _OFF_

### Expected Results/How It Should Work:
With the knob correctly aligned on the rotary switch, the arrow should point to either *On* or *Off* with the rotation of travel (90 degrees) from the 9-o-clock to the 12-o-clock position, by way of the 10-o-clock.  

### Actual Results/How It Does Work:

[image]

In this configuration with the rotary switch mounted to the backplate and the indexing pin (*#1*) at the 12-o-clock position, the shaft is currently rotated all the way to the **LEFT** 90 degrees.  This results in the flat edge of the shaft (*#2*) facing the 12-o-clock position as well.  

[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (Fusion 360)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [X] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):

*OH4A7A2-1 - ASSY, OBOGS PANEL.*

Exprezz0, 2022-11-30:
Fixed Model OH4A7A2-1 - ASSY, OBOGS PANEL

Process:
1:  Deleted Rigid Joint.  Rotated Rotary Switch 90* to the left.
2:  Filled hole in *OH4A7A2-11 - OBOGS BACKPLATE* that was left behind from indexing pin on rotary switch.
3:  Cut new hole (.125) in *OH4A7A2-11 - OBOGS BACKPLATE* to facilitate new position of indexing pin.
4:  Added new Rigid Joint
5:  Dissolved Extrudes/Deleted Sketches from Browser 
6.  Saved User Version with Remarks. (v19 --> v20)
7. 
[image]

OBOGS Knob will now point to "OFF" position in model, but will function properly once produced.

jrsteensen, 2023-03-25:
Closed per above! Thanks @Exprezz0

closed 2023-03-25 by jrsteensen