jrsteensen/OpenHornet #599: Master Arm Module Spacer: Hole missing

Higgins-OH, 2022-12-04:
### Summary:
One hole for the screw is missing in the Master Arm Module Spacer

### Expected Results/How It Should Work:
Needs hole for the middle screw.

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Upper Instrument Panel (UIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH1A2-14

Lazarus94, 2023-03-26:
Fixed in OH1A2-1 - ASSY, PANEL, MASTER ARM V13 and OH1A2-1_MOD_A - ASSY, PANEL, MASTER ARM V6

jrsteensen, 2023-03-26:
### Review Comments

- [ ] Need to update up through the console level - should be fixed in UIP vXXXX with latest components.

jrsteensen, 2023-03-26:
Fixed in UIP v68

closed 2023-03-26 by jrsteensen