jrsteensen/OpenHornet #603: SNSR engraving panel: radial cutouts for backlight missing

Higgins-OH, 2022-12-17:
### Summary:
The knobs are supposed to be backlit, the engraving is missing at the RADAR and INS holes.

### Screenshots:

[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [x] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH5A7A1-12

kbastronomics, 2023-03-25:
Fixed in OH5A7A1-1 - ASSY, SNSR PANEL v18

jrsteensen, 2023-03-25:
### Review Results:

- [ ] Please ensure you update all assemblies up to the console level with the latest version.

[image]

jrsteensen, 2023-03-26:
Closed in RC v114

closed 2023-03-26 by jrsteensen