jrsteensen/OpenHornet #604: Two alternative brightness knobs included with the AMPCD Fusion 360 assembly

kika, 2022-12-21:
### Summary:

AMPCD Fusion 360 assembly contains two brightness knobs on top of each other. One is hidden, one is shown by default, but if you turn them on/off frequently (which is necessary when working with such a complex assembly) it's easy to mix them up. One is compatible with the brightness potentiometer and another uses ¼" shaft and can't be used with the current pot.
  * AMPCD MS21382-0XNX (Modified) this is the good one
  * OHC-KNB-MS21382-12 - KNOB, GREY, .25 D-SHAFT this is the wrong one, at least for the given potentiometer

### Expected Results/How It Should Work:
Only one part is included in the final assembly for AMPCD

kbastronomics, 2023-03-25:
Fixed in Fusion File OH2A3A1-1 - ASSY, AMPCD MODULE V11

jrsteensen, 2023-03-25:
### Review Results:
- In the OH>Common>OHC0003 - Knobs folder:
  - [ ] Move correct knob there.
  - [ ] Add it to the spreadsheet.
  - [ ] Rename the knob.
  - [ ] Add the knob to the OHC0003 Collector assembly.

kbastronomics, 2023-03-26:
Updated OH2A1-1 - ASSY, LOWER INSTRUMENT PANEL V99
Updated OHC0003 COLLECTOR ASSY V6

jrsteensen, 2023-03-26:
Fixed in LIP v101, Knob Collector v7.

closed 2023-03-26 by jrsteensen