jrsteensen/OpenHornet #605: Incorrect switches (2x) used on OH1A7-1 - ASSY, PANEL, HUD v5

Exprezz0, 2022-12-21:
### Summary:
Referencing the B1HWF model, the DAY/NIGHT and ALT select switches are displaying as `100SP3T8B13M1QEH - Toggle Switches SPDT ON-OFF-ON SLDR` on the HUD Panel Assembly.  

### Expected Results/How It Should Work:
The Day/Night Hud selector only has two options.  Same goes for the Altimeter selection switch (BARO and RDR).
Both Switches should be `100SP1T8B13M1QEH - Toggle Switches SPDT ON-ON SLDR MNT`.  This switch is in keeping with the same characteristics and dimensions as the current switch with the only difference being extra "OFF" position.

### Screenshots:
[image]

## More Information
### Category:
*Check one or more items.*
- [x] MCAD (Fusion 360)

### Type:
*Check one item.*
- [x] Bug

### Applicable End Item:
*Check one item.*
- [x] Main Instrument Panel (MIP)

### Associated Filename(s):
OH1A7-1 - ASSY, PANEL, HUD v5

Lazarus94, 2023-03-26:
Fixed in OH1A7-1 - ASSY, PANEL, HUD V6.

jrsteensen, 2023-03-26:
### Review Comments

- [ ] Need to update up through the console level - should be fixed in UIP vXXXX with latest components.

jrsteensen, 2023-03-26:
Fixed in UIP v69

closed 2023-03-26 by jrsteensen