jrsteensen/OpenHornet #610: Spin Panel: BRT marker missing?

Higgins-OH, 2022-12-30:
### Summary:
The HMD Brightness doesn't have a "BRT" marker, unlike the engravings for the AOA on the HUD panel.
Sources are not consistent, so this is more a question than a definite issue. Shouldn't there be a marker of some sort?

### Screenshots:
[image]

## More Information
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [x] Upper Instrument Panel (UIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH1A6-12 Spin Panel Legend Plate

kbastronomics, 2022-12-30:
While not the perfect reference it does have the marking in DCS
[image]

Higgins-OH, 2022-12-30:
Well, the DCS panel looks different than the OH panel - different versions or missing lines?
[image]

jrsteensen, 2023-01-03:
We are using a different version, more accurate to the Block that we are representing. I believe this is correct as-is.

closed 2023-01-03 by jrsteensen