jrsteensen/OpenHornet #612: LDG Gear Panel: Engraving wrong colour

Higgins-OH, 2022-12-30:
### Summary:
Panel only showing yellow and black engraving, there should also be white.

### Expected Results/How It Should Work:
According to various sources there aren't only yellow stripes, but white/yellow alternating like on the other panels.

### Actual Results/How It Does Work:
Stripes in black and yellow only.

### Screenshots:
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [x] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*
OH4A2A1-13 Landing Gear Panel Legend Plate

Exprezz0, 2023-01-03:
The lines are already sketched and separated on the panel, but they weren't filled with the appropriate color.  The only modification that needs to happen to fix this issue is application of color using the Appearance menu option (see attached):

[image]

jrsteensen, 2023-01-03:
Please feel free @Exprezz0!

Exprezz0, 2023-01-03:
Legend Panel fixed.  File saved on the cloud with description (v20).

jrsteensen, 2023-01-03:
Thanks @Exprezz0!

closed 2023-01-03 by jrsteensen