jrsteensen/OpenHornet #614: AMPCD Controller IRQ PIN not Connected to IRQ Capable PIN on Pro-Micro

kbastronomics, 2023-01-02:
### Summary:
The AMPCD Controller board is designed to use the IRQ pin of the TCA9554PWR to let us know when a switch is pressed,   the PIN on the PRO-Micro though it's connected to is not IRQ Capable.

### Expected Results/How It Should Work:
IRQ routine would be written so that when the any button is pressed a Interrupt is triggered and the software can read the TCA9554 to see which one it was.

### Actual Results/How It Does Work:
Since the D6/A7 Pin is not IRQ capable you can not use a interrupt route and must poll the switches.
**Swapping PINS D6 with D8 would resolve this issue which also allows use of Pin Change Capture IRQ there would be no impact functionality but is a board change**

### Screenshots:

From the AMPCD COntroller showing the IRQ Pin.

[image]

From the PIN OUT of the PRO-Micro show that D6/A7 is not a IRQ capable pin

[image]

And from the datasheet

[image]

## More Information
### Category:
*Check one or more items.*
- [X] ECAD (PCB Design or other electrical hardware)
### Type:
*Check one item.*
- [X] Bug
### Applicable End Item:
*Check one item.*
- [X] Lower Instrument Panel (LIP)
- [X] Main Instrument Panel (MIP)

### Associated Filename(s):
ddi_ampcd_controller

jrsteensen, 2023-01-02:
Is this a breaking bug that prevent the AMPCD from working at all, or is it an enhancement to reduce overhead on the AMPCD?

kbastronomics, 2023-01-02:
This prevents it from working as designed.
It can be coded around.
   
The intent of the IRQ pin was to prevent the need to use polling of the buttons and instead the IRQ pins would let you know a button was pressed then you go check for which one.

Instead we'll have to POLL which means constantly checking for a button press and if found update DCS-BIOS

jrsteensen, 2023-01-02:
Perfect, thank you. I'll save this one for the next major revision of the PCB in the future then. :)

Arribe, 2024-06-11:
The current code's work around is to set IRQ pinmode to OUTPUT, which allows users to connect the IRQ pin according to the interconnect.  An alternative workaround is to disconnect the IRQ pin connection on the controller PCB, then one could set the IRQ pinmode to INPUT.

Arribe, 2024-06-12:
Aside from the UFC using an ESP32, the sample code from scuba also attaches the IRQ pin to use interrupt logic.