jrsteensen/OpenHornet #617: DDI brightness selector/power switch is wrong component

DeadMeatDCS, 2023-01-09:
### Summary:
Wrong rotary switch component for DDI brightness selector/power switch. 
Current component is a switch with an index angle of 60 degrees. Should have been a switch with a 45 degree index angle to match legend and reference.

### Expected Results/How It Should Work:
Legend marks in model and in reference pictures indicate 45 degrees switch steps, calling for a rotary switch with the same index angle.

Appropriate switch would be a **MRX108** (2-8 positions, 45 degree index angle) if we want stick with the same NKK family as the _current MRY106_. It has a few differences however so a small update is necessary to the DDI mounting spacer:

- Keyway and locking ring sit at a different angle to the flat on the switch shaft, so the hole for that needs to be moved (see below).
- _Diameter is the same, but the switch bushing is 0.004" taller. It should stay clear of the DDI PCB though._
- _Switch body and common terminal is longer, adding 0.095" to the length but there's nothing behind the DDI, so not a problem._

### Actual Results/How It Does Work:
Current switch in the model is a MRY106, which has 60 degree index angle (2-6 positions), not matching legend and reference.

### Screenshots:
Reference
[image]

Legend in model
[image]

Anti rotation hole to be moved
[image]

Current switch
[image]

Suggested switch
[image]

## More Information
### Category:
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [x] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH1A3-18 - DDI MOUNTING SPACER
MRY106 - SWITCH, ROTARY, 2-6 POS, NKK

jrsteensen, 2023-03-26:
Thanks for the write up @DeadMeatDCS! This is an example of an excellent issue writeup!

jrsteensen, 2023-03-26:
Fixed in UIP v71

closed 2023-03-26 by jrsteensen