jrsteensen/OpenHornet #622: Spin Panel/HMD Poti: Not enough threads for PCB and backplate

Higgins-OH, 2023-01-30:
The poti's threads are not enough to fasten it with the nut on the backplate. In the model this problem is not visible because the margin between poti and backplate is too small, it clips with the PCB.

If there were a coutout in the backplate big enough for the nut it would work - but then the washer might cause a short circuit on the nearby LEDs and should be out of plastic.

### Screenshots:
[image]
## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [x] UIP   
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH1A6-10 - Spin Panel Backplate
P170S-FC20BR10K Pot (HMD)

jrsteensen, 2023-03-25:
For this one, lets go with the enlarged backplate hole and printed plastic washer.

jrsteensen, 2023-03-27:
Fixed in LIP v71

closed 2023-03-27 by jrsteensen