jrsteensen/OpenHornet #623: Video Record Panel/IFEI pot: Not enough space in backplate for nut

Higgins-OH, 2023-02-06:
### Summary:
In order to fasten the pot to the PCB, the hole in the backplate needs to be bigger.

### Expected Results/How It Should Work:
The backplate should have a hole big enough for the pot's nut to fit.

### Actual Results/How It Does Work:
No possibility to fasten pot: No hole for nut/threads on pot not enough.

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [x] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH2A2A1A3-10 Hud Video Record Panel Backplate; P170S-FC20BR10K pot (brightness)

jrsteensen, 2023-03-25:
Evaluate if different pot should be used here with more thread, or if we can use the big backplate hole with a thin plastic washer.

jrsteensen, 2023-03-27:
Fixed in LIP v102

closed 2023-03-27 by jrsteensen