jrsteensen/OpenHornet #624: OH4A3A1-100 - SERVOCITY 24T, 0.250" (1/4) Bore 32P Shaft Mount Pinion Gear DISCONTINUED

ajmilner85, 2023-02-10:
### Summary:
Subject component appears to be discontinued by SERVOCITY.
https://www.servocity.com/24t-0-250-1-4-bore-32p-shaft-mount-pinion-gear/

SC recommends part https://www.servocity.com/2303-series-steel-mod-0.8-pinion-gear-1/4-round-bore-24-tooth-/ as an alternative. Dimensions seem ok, not sure if material matters. Alternative is zinc-plated steel vs brass.

### Expected Results/How It Should Work:
N/A

### Actual Results/How It Does Work:
N/A

### Screenshots:
N/A

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ X ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ X ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ X ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH4A3A1-100

ajmilner85, 2023-03-04:
For Rudder Trim and Select Jett Mechanism The Upper Bracket faces that mate to the Lower Mount Plates need to be offset an additional 0.052" to accommodate the [2303 Series Steel Pinion Gear](https://www.servocity.com/2303-series-steel-mod-0.8-pinion-gear-1/4-round-bore-24-tooth-/). Otherwise the gear mesh together fine.

[image]

For the Select Jett Mechanism, the Rotating Shaft (OH4A3A1-105) may also need to be lengthened. Will confirm once I've printed the rest of the assembly.

jrsteensen, 2023-03-26:
Fixed FCS in LC v133
Fixed Select Jett in LC v135

closed 2023-03-26 by jrsteensen