jrsteensen/OpenHornet #625: AV COOL, PLacement of CAP vs Switch does not allow Switch to be positioned correctly

kbastronomics, 2023-02-12:
### Summary:
By the Fusion360 file there is no 1000uF CAP on the PCB for the AV Cool Switch but the ECAD does include the CAP.  with the CAP soldered into position the S1AL switch will not sit correctly as the body of the switch hits the CAP.
This causes the switch to be about 5 decrees rotated from the correct orientation

NOTE:  the CAP could be removed and not impact function on this board as a simple fix 

### Expected Results/How It Should Work:
Switch should be 90 degree's from the bottom edge of the panel,   

### Actual Results/How It Does Work:
Switch sits about 85 degree's from the bottom edge of the panel giving the switch a slight rotation 

### Screenshots:
Fusion Model vs actual assembled panel

[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [X] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH5A2A3-1 - ASSY, AV COOL SWITCH

jrsteensen, 2023-03-25:
Cap should be omitted. Change to a "NI" (Not Installed) component in ECAD.

Exprezz0, 2023-04-22:
Fixed with commit: 1bbc288c26f1a995a60497a069b2bf99ec55dfcf.  Will be implemented with #592.

PCB Step file has been uploaded to F360, but not implemented into the design yet.  Capacitor was placed towards the top, which should allow for free movement of NKK toggle switch.

jrsteensen, 2023-04-22:
[image]
[image]

Still interferes. Should be marked as Do Not Populate. (https://www.kicad.org/blog/2023/02/Version-7.0.0-Released/, about halfway down.) Will suppress cap from PCB model. Want to keep it on the schematic so if there is an application for the PCB not in the OH frame, it can be installed.

Exprezz0, 2023-04-22:
Fixed and closed.  Move capacitor to prevent collision with supports.  Updated with commit: be390c8712f30efc8076182859b651b58c88dd0b

closed 2023-04-22 by Exprezz0