jrsteensen/OpenHornet #629: Comm Panel Backplate: Cutout for ready LED pins missing

Higgins-OH, 2023-02-19:
The backplate lacks a cutout for the solder pins of the ready LED.

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [x] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH4A7A1-11 Comm Panel Backplate

Arribe, 2023-03-25:
Agree with the poti nut cut out. I did a recut to increase the hole diameter to get it to fit. I do not have the ready LEDs yet.

jrsteensen, 2023-03-25:
Fixed in LC v129

closed 2023-03-25 by jrsteensen