jrsteensen/OpenHornet #636: OH4A1A1-49 - LEFT AUX SUPPORT BRACKET Interference with OH4A1A1-45 LEFT AUX PANEL Mounting Hardware

ajmilner85, 2023-03-01:
### Summary:
The bottom right #6-32 X 0.5" pan head screw used to mount the LEFT AUX PANEL intersects with the LEFT AUX SUPPORT BRACKET body. Suggest adding a recess in the LEFT AUX SUPPORT BRACKET to accommodate the full length of the 0.5" screw or designate shorter hardware.

### Screenshots:
[image]
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [X] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH4A1A1-49 LEFT AUX SUPPORT BRACKET

jrsteensen, 2023-03-26:
[image]

jrsteensen, 2023-03-26:
Fixed in LC v138

closed 2023-03-26 by jrsteensen