jrsteensen/OpenHornet #637: RWR control: rotary switch nut missing, legend plate cutout too small for standard nut

Higgins-OH, 2023-03-04:
### Summary:
The nut of the rotary switch is missing in the plans, therefore it isn't obvious the standard nut doesn't fit the cutout.

### Expected Results/How It Should Work:
Either we need to find a smaller, non-standard nut or the cutout needs to be larger for it to fit.

### Actual Results/How It Does Work:
At the moment the switch's standard nut doesn't fit the cutout.

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [x] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH2A4A1-11 RWR control panel legend plate

jrsteensen, 2023-03-26:
@Lazarus64 is claiming this one.

Lazarus94, 2023-03-26:
Fixed in LIP V98

jrsteensen, 2023-03-26:
Small correction: This is how the file properties should look - only a part number in the part number field, only description in the description field, and combined/separated in a dash in the part name.
[image]

Fixed in LIP v99

closed 2023-03-26 by jrsteensen