jrsteensen/OpenHornet #643: HUD Panel missing engravings for backlighting of knobs

kbastronomics, 2023-03-27:
### Summary:
All 4 Knobs on the HUD panel should be backlit but there is no engraving to allow backlight to come thru

### Expected Results/How It Should Work:
Knobs should be backlit.

### Actual Results/How It Does Work:
no engraving to allow backlight to come thru prevents knobs from be lit up
### Screenshots:

[image]
[image]
[image]

## More Information

### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library

### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation

### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [X] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
- [ ] 
### Associated Filename(s):
OH1A7-1 - ASSY, PANEL, HUD

kbastronomics, 2023-03-28:
Fixed in UIP v75

jrsteensen, 2023-03-28:
Closed with Sandra's fix.

closed 2023-03-28 by jrsteensen