jrsteensen/OpenHornet #644: UFC v6

jrsteensen, 2023-03-27:
## Justification
The UFC v5 IO Expander (AW9523BTQR) IC is End Of Life (EOL). That said, there are about 25k of them still available so no pressing need for a redesign for Beta1.

## Requirements
- [ ] #595
  - This is not a breaking issue, since we install the displays with a standoff from the actual PCB.
- [ ] #596
  - This issue may or may not be a big deal - I remember looking at this at design time and thinking we would be able to get all the required UFC F18 characters as-is, but I could be mistaken.
- [ ] Find a suitable replacement for AW9523BTQR IO Expander with a similar form factor and capability.
- [ ] Require no UFC MCAD changes (drop-in replacement)

@kbastronomics please chime in as necessary so we can start hammering out requirements.

### Timeline: 
Prior to EOL/Post Beta1 Release

## Dependencies
#592 is a blocking dependency.

jrsteensen, 2023-04-05:
Evaluate the following IO Expanders for use with the UFC:
https://jlcpcb.com/partdetail/TexasInstruments-TCA8418RTWR/C138713
https://jlcpcb.com/partdetail/TexasInstruments-TCA6408ARSVR/C2649390
https://jlcpcb.com/partdetail/TexasInstruments-PCA9555RGER/C2652275
https://jlcpcb.com/partdetail/TexasInstruments-TCA6416ARTWR/C182336
https://jlcpcb.com/partdetail/Onsemi-FXL6408UMX/C457770
https://jlcpcb.com/partdetail/TexasInstruments-PCF8574RGTR/C2652270
https://jlcpcb.com/partdetail/NxpSemicon-PCA9557BS118/C2652288

kbastronomics, 2023-04-05:
TCA8418RTWR supports keyboard  matrix with 18 IO in hardware which is perfect for the UFC.
the UFC is basically a DSKY (if you know the reference) having key scanning in hardware vs doing thru bit banging a IO multplex will make for more reliable code from my POV..   we can drive the backlight from the processor.

a 5x4 matrix will cover the UFC itself,   you can then add either the AP keys directly or put a TCA6408ARSVR on the AP board

jrsteensen, 2023-04-05:
> TCA8418RTWR supports keyboard matrix with 18 IO in hardware which is perfect for the UFC. the UFC is basically a DSKY (if you know the reference) having key scanning in hardware vs doing thru bit [...]

The only problem with the TCA, is that its 3.3v and not 5v tolerant. So now I have to run a 3.3v and 5v i2c network, and complicates routing on the board if I locate the IC near the micro instead of in the 5V power fill zones.

I'm not entirely sure if I get your bitbanging issue - there are Arduino libraries already existing for these IO expanders that make them pretty transparent from a programming standpoint?

jrsteensen, 2023-04-06:
[UFC_Main.zip](https://github.com/jrsteensen/OpenHornet/files/11172411/UFC_Main.zip)
For schematic review

jrsteensen, 2023-04-07:
Redlines implemented as discussed, PCB routed. For review.

[UFC_Main-CHK2-Routed.zip](https://github.com/jrsteensen/OpenHornet/files/11175125/UFC_Main-CHK2-Routed.zip)

kbastronomics, 2023-04-07:
So far looking good,
**Encourage anyone with PCB experience to take a look**

jrsteensen, 2023-04-07:
Yes, please. Any and all eyes we can get on this the better.

jrsteensen, 2023-04-07:
[image]
[image]

- [ ] Verify, and likely fix and reroute footprint for TCA. Verify all other footprints.

jrsteensen, 2023-04-07:
Its correct as is. Its on the back face, rotated 180*. Got a little ahead of myself at 0100 local.

jrsteensen, 2023-04-07:
Production candidate files

[UFC_Main.zip](https://github.com/jrsteensen/OpenHornet/files/11181731/UFC_Main.zip)
[UFC_Function Select.zip](https://github.com/jrsteensen/OpenHornet/files/11181739/UFC_Function.Select.zip)

jrsteensen, 2023-04-08:
Prototype PCBs ordered. Considering this closed.

closed 2023-04-08 by jrsteensen