jrsteensen/OpenHornet #646: Integrate latest SEL JETT PCB to Left AUX Console

jrsteensen, 2023-03-28:
Get and integrate the latest SEL JETT PCB into the SEL JETT ASSY and Left AUX Console.

jrsteensen, 2023-04-02:
This issue should also include #657 fix

Exprezz0, 2023-04-16:
Left Aux Panel updated to receive PCB Revision 4.  
-inlay adjusted to fit PCB, specifically in top left corner where PCB was exposed and collided with Left Rib (A?)
-8x32 PHS mounting screw in bottom right corner (hidden by PCB) countersunk to prevent collision with PCB.
-Hole added to accept Ready LED on PCB (closed in different issue)
-Left Aux Panel interior wall adjusted (.1-.2in) to close exposed gap from PCB/Backlight panel.  Doesn't collide or prevent SEL JETT ASSY from connecting.

[image]

[image]

Should be ready for #514 once reviewed.

closed 2023-04-16 by Exprezz0