jrsteensen/OpenHornet #647: OH4A2A1-1 Landing Gear Panel legend and light plate do not cover back plate cutout  to PCB at bottom center

Arribe, 2023-03-28:
### Summary:
The light and legend plates do not extend over a backplate cutout on the landing gear panel.  LED backlighing for the DN legend will likely blead out the hole.

### Expected Results/How It Should Work:
The cut out for the LED/Resistor should be covered by the light and legend plate.

### Actual Results/How It Does Work:
There is a small gap that doesn't cover the backplate cutout fully.

### Screenshots:

[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [x] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

Arribe, 2023-03-28:
I have this gap in the panel I built, and have seen photos on discord from other builders that also have this gap.

jrsteensen, 2023-03-28:
As a workaround, pot it with black RTV or similar. 

We will get this in after B1 - trying very hard to not change any PCBs for relatively minor things until there is something that drives us to do more than one thing to a PCB.

closed 2026-09-28 by jrsteensen