jrsteensen/OpenHornet #651: Create Drawing: OH4A10-1 - ASSY, TRANSDUCER GAIN REMOTE

Lazarus94, 2023-04-01:
Create Drawing: OH4A10-1 - ASSY, TRANSDUCER GAIN REMOTE

Lazarus94, 2023-04-01:
Add to project OpenHornet v1.0.0-beta1 Release

Lazarus94, 2023-04-01:
IN CHECK

[OH4A10-1 - ASSY, TRANSDUCER GAIN REMOTE v1.pdf](https://github.com/jrsteensen/OpenHornet/files/11129690/OH4A10-1.-.ASSY.TRANSDUCER.GAIN.REMOTE.v1.pdf)

jrsteensen, 2023-04-01:
IN BACKCHECK
[image]

jrsteensen, 2023-04-02:
Incorporated backcheck comments, added to RC release package, and closed.

[OH4A10-1 - ASSY, TRANSDUCER GAIN REMOTE_-_v3.pdf](https://github.com/jrsteensen/OpenHornet/files/11132973/OH4A10-1.-.ASSY.TRANSDUCER.GAIN.REMOTE_-_v3.pdf)

closed 2023-04-02 by jrsteensen