jrsteensen/OpenHornet #657: Need cutout added to LEFT AUX PANEL to allow LED to pass thru

kbastronomics, 2023-04-02:
### Summary:
Left AUX Panel does not have cutout for LED to pass thru.   Needs cutout added to OH4A1A1-45 - LEFT AUX PANEL

### Expected Results/How It Should Work:
LED should pass thru and be visable

### Actual Results/How It Does Work:
WIth LED in place you can not seat the SELECT JETT Panel properly

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [X] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH4A1A1-1 - ASSY, LEFT CONSOLE STRUCTURE
OH4A1A1-45 - LEFT AUX PANEL

jrsteensen, 2023-04-02:
Should be completed with #646

Exprezz0, 2023-04-16:
LEFT AUX PANEL (MCAD) has been corrected to allow for LED when the PCB is placed in position.  Fixed in OH4A1A1-1 - ASSY, LEFT CONSOLE STRUCTURE v35+

[image]

closed 2023-04-16 by Exprezz0