jrsteensen/OpenHornet #660: Throttle Grip Holes for 98381A471 Pins Too Small

Arribe, 2023-04-04:
### Summary:
The holes for the pins between both of the 2 throttle grip halves are too narrow.
### Expected Results/How It Should Work:
The pins should go in a little easier to avoid cracking the resin grips.  

### Actual Results/How It Does Work:
The holes are significantly narrower than the pins and close to grip faces/edges.  Even with trying to scrape them larger to push in the pins the odds of cracking the grip is high.  I broke one and chipped a second reprint.  The one below the TDC joystick is especially problematic.

### Screenshots:

[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [x] Throttle
- [ ] General Software
### Associated Filename(s):
OH4A9-106 - ASSY, INBOARD THROTTLE GRIP
OH4A9-205 - ASSY, OUTBOARD THROTTLE GRIP

kbastronomics, 2023-04-04:
Just a note,  I did not have this issue on my resin prints ,  mine fit nice and snug, no play.  just a tiny bit of elephant foot due to my printing with the joining sides on the build plate.

Arribe, 2023-04-04:
Reverified part number that I'm using matches the design.  I printed the grips vertically, so if that is what's causing the issue then it could be covered with manufacturing notes.  One can also omit problematic pins when assembling the grips.

kbastronomics, 2023-04-04:
if you printed vertically there is a tendency even with supports for holes to droop   ie not be round.  maybe that's what you're running into.

jrsteensen, 2023-04-17:
Expanded clearance holes slightly, should result in a free running fit for the pins. So that way, you get alignment but doesn't have to press into the resin. Resolved in TQ v22

closed 2023-04-17 by jrsteensen