jrsteensen/OpenHornet #670: OBOGS - Interconnect Diagram for Oxy Flow Knob A11405RNZQ needs 2 data  pins

Arribe, 2023-04-08:
### Summary:
DCSBios has 2 commands for OXY Flow Knob, one if using a pot another if using an encoder.  A11405RNZQ  is a 2 position rotary encoder.  It needs 2 pins to integrate with DCS BIOS code.

### Expected Results/How It Should Work:
Rotating the Oxygen flow knob should rotate the knob in DCSBios / Game.

### Actual Results/How It Does Work:
Since the interconnect diagram only calls for one pin to be used in addition to ground, the BIOS command:
DcsBios::RotaryEncoder oxyFlow("OXY_FLOW", "-3200", "+3200", PIN_A, PIN_B); does not have enough pins to trigger off and on action.

My working code snippet showing both pins in addition to ground need to be connected.  Note, I chose 15 because it was open and easy to get to for proof-of-concept testing.

const int oxyFlowOn = 2;
const int oxyFlowOff = 15;
DcsBios::RotaryEncoder oxyFlow("OXY_FLOW", "-65535", "+65535", oxyFlowOff, oxyFlowOn);

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [x] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [x] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
Interconnect Diagram: 4A7A2 ABSIS_ALE - 4A7A2SW2 OXY FLOW SW

jrsteensen, 2023-04-08:
I'm not super proficient on the programming side, but why are you defining it as a rotary encoder instead of a switch?

Arribe, 2023-04-08:
I'm following the code reference tool that shows 2 options below.  I just tried coding it as DcsBios::Switch2Pos oxyFlow("OXY_FLOW", oxyFlowOn); to try to match the interconnect diagram but that didn't work.  Rotary encoder with 2 pins works.  Only other option appears to be a 90 degree pot then that would be 1 pin.
 
[image]

Arribe, 2023-04-08:
Since it's not a free spinning rotary encoder, that's why the -65535, +65535 values need to be set in the code.  It's either full on or full off in one step.

DcsBios::RotaryEncoder oxyFlow("OXY_FLOW", "-65535", "+65535", oxyFlowOff, oxyFlowOn);

DeadMeatDCS, 2023-04-14:
Late to the party here, but you absolutely can control the OXY_FLOW knob with just 1 digital pin and a 2-position rotary switch connected like a OFF-ON toggle switch (like the interconnect shows).

Just do a digitalRead of the pin and do the appropriate sendDcsBiosMessage:

```
/*
  Tell DCS-BIOS to use a serial connection and use the default Arduino Serial
  library. This will work on the vast majority of Arduino-compatible boards,
  but you can get corrupted data if you have too many or too slow outputs
  (e.g. when you have multiple character displays), because the receive
  buffer can fill up if the sketch spends too much time updating them.
  
  If you can, use the IRQ Serial connection instead.
*/
#define DCSBIOS_DEFAULT_SERIAL

#include "DcsBios.h"

#define OXY_FLOW_PIN 2 //ON position connected to this pin

int read = 0;
int read_ = 0;

/* paste code snippets from the reference documentation here */

void setup() {
  DcsBios::setup();
  pinMode(OXY_FLOW_PIN, INPUT_PULLUP);
}

void loop() {
  DcsBios::loop();

  read = digitalRead(OXY_FLOW_PIN);
  
  if (read != read_){ //Has switch been toggled?
    read_ = read;
    if (read == LOW){ //Switch has been turned on
      sendDcsBiosMessage("OXY_FLOW", "65535"); //Tell DCS-BIOS to turn the knob all the way on
    }
    else{
      sendDcsBiosMessage("OXY_FLOW", "0"); //Turn it all the way off
    }
    
  }

}
```

jrsteensen, 2023-04-16:
Will implement the easy software solution above versus a hardware change.

closed 2023-04-16 by jrsteensen