jrsteensen/OpenHornet #675: OH4A7A-11 ANT SEL - Should have 3 positon switches

Arribe, 2023-04-09:
### Summary:
In DCS the ANT SEL panel uses 3 position switches: comm 1 = lower, auto, upper; IFF = lower, both, upper

### Expected Results/How It Should Work:
To fully control the pit the switches should be replaced with the 3-position version.  The interconnect diagram will also need to be updated to account for connecting each switch to 2 pins each on the mega.

### Actual Results/How It Does Work:
The design uses 2 position switches so cannot easily control all the available positions in DCS.

### Screenshots:

[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [x] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [x] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH4A7A-11 ANT SEL - COMM1 and IFF switches.

jrsteensen, 2023-04-09:
Verified, and should be incorporated.

- [ ] Update model
- [ ] Update drawing

Exprezz0, 2023-04-10:
ANT SEL PCB has been updated and will be pushed with #592 PR.
Back Plate needs to be cutout still for F360 Model.

jrsteensen, 2023-04-11:
Great team effort there, thanks @Exprezz0! Closed with latest commit. (PCB native files will be brought in with #592.)

commit 35b3a88d (2023-04-11): Committing more drawings.

Closes #675 and #491
closed 2023-04-11 by jrsteensen