jrsteensen/OpenHornet #677: Throttle Solenoid linkage dowel diameter too large

Arribe, 2023-04-11:
### Summary:
The 91595A078 dowel pin diameter is too large for the DSTL-0840-05 solenoid hole.

### Expected Results/How It Should Work:
The dowel pin should fit into the solenoid to attach the linkage and hold the spring retainer.

### Actual Results/How It Does Work:
The dowel pin has a 2.5mm diameter, but the Solenoid spec is only a 2.4mm opening +-0.05.  The pin doesn't fit.  I've placed an order for the 2mm diameter version of the pin 91595A034.

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [x] Throttle
- [ ] General Software
### Associated Filename(s):
OH49-16 - Solenoid Linkage

jrsteensen, 2023-04-12:
I'm thinking the [91585A240](https://www.mcmaster.com/91585A240/) will work just as well, and is 4 dollars cheaper.

jrsteensen, 2023-04-12:
Fixed in Throttle v21. Reprint on the retainers optional, since the pins are retained laterally by the body plates.