jrsteensen/OpenHornet #679: Silkscreen Error HOTAS_Analog Stick Breakout PCB

Arribe, 2023-04-11:
### Summary:
The silkscreen for the PCB has 2 pads labled as Y.  One should be marked as X to make wiring easier and to reflect which pad is X vs Y.

### Expected Results/How It Should Work:
The pads are labled so one can determine which pad to wire to which connector pin without having to open KiCad to determine the associated pad.

### Actual Results/How It Does Work:
2 Pads are labled as Y, including the one that's marked to be connected to J3 of the connector.

### Screenshots:
[image]

[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [x] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [x] Throttle
- [ ] General Software
### Associated Filename(s):
HOTAS_Analog Stick Breakout.kicad_pcb

Exprezz0, 2023-04-16:
Fixed.  Will be implemented with #592 
[image]

kbastronomics, 2023-08-25:
ECAD Not Updated,   re-opening

jrsteensen, 2023-08-25:
Fixed in e634feb03b69f75de585a6db4d460b8a841cce22

commit e634feb0 (2023-08-25): Updated HOTAS_Analog Stick Breakout PCB

Silkscreen Changes:
- Changed Y on center top pad to X
- Changed BUT to SW for consistency.
Updated manufacturing files.
Closes #679