jrsteensen/OpenHornet #680: APU ENG CRANK PANEL - LED violates JLC constraints

Exprezz0, 2023-04-13:
### Summary:
The placement of an LED located on the APU ENG CRANK Panel, specifically the superier LED that lights up "L" for the engine crank switch falls within the boundaries of one of the mounting holes for the panel and the switch itself. 

### Expected Results/How It Should Work:
1:  LED should be placed on the PCBs in order to meet JLC production requirements, despite it making through assembly in previous revisions.
2:  LED should be offset and not placed directly behind engraving on the legend panel.  
3:  LED should be placed in a matter that allows complete connection.

### Actual Results/How It Does Work:
[image]
Due to board constraints, the +5v pad and GND pad over lap the keepout zone for both the 6x32 mounting screw hole and mounting hole for 631H toggle switch.  LED needs to me moved, removed, or rotated in order to fit within the constraints.  Unable to currently get +5v power to the LED without moving.

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [x] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [x] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
APU ENG CRANK PANEL

Exprezz0, 2023-04-13:
- [ ] PCB Design (Silkscreen, DRC)
- [ ] Schematic
- [ ] JLC Manufacturing Files
- [ ] STEP file
- [ ] PNGs
- [ ] Back Plate (MCAD)

kbastronomics, 2023-04-13:
I think the constraint should only be on the bottom side of the board not the top where the components are.   
I know this is possible in other ecad software, not sure about kiCAD though

Exprezz0, 2023-04-13:
APU ENG CRANK Panel has been fixed.  Saved and committed as:  https://github.com/jrsteensen/OpenHornet/pull/592/commits/408c4cbd24c7802eabae8e6bfb4a1d68226126a0

PCB Model has been added and updated to F360, renamed to updated nomenclature, and backplate adjusted to accept new PCB.  Saved as v18.

Mechanical files/Drawings are ready to be updated.

closed 2023-04-15 by Exprezz0