jrsteensen/OpenHornet #682: AV COOL minifit connectors detached in model

Lazarus94, 2023-04-13:
### Summary:
The PCB model is broken. 

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH5A2A3-1 - ASSY, AV COOL SWITCH

Higgins-OH, 2023-04-13:
Isn't that thought as a reminder to use pigtails, like with the landing gear PCB?

jrsteensen, 2023-04-13:
Its an indicator to pigtail the connectors. Model is not broken. Correct as-is, and pigtails should be called out on the drawing.

closed 2023-04-13 by jrsteensen