jrsteensen/OpenHornet #688: Create Drawing: OH4A6A1-100 - ASSY, RUDDER TRIM MECHANISM

Lazarus94, 2023-04-14:
Create Drawing: OH4A6A1-100 - ASSY, RUDDER TRIM MECHANISM

Lazarus94, 2023-04-14:
Closed as can be incorporated into FCS panel drawing.

Lazarus94, 2023-04-15:
IN CHECK

[OH4A6A1-100 - RUDDER TRIM MECHANISM  v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11239572/OH4A6A1-100.-.RUDDER.TRIM.MECHANISM.v4.pdf)

jrsteensen, 2023-04-16:
IN BACKCHECK

[OH4A6A1-100 - RUDDER TRIM MECHANISM_-_v4-CHK1.pdf](https://github.com/jrsteensen/OpenHornet/files/11243069/OH4A6A1-100.-.RUDDER.TRIM.MECHANISM_-_v4-CHK1.pdf)

Lazarus94, 2023-04-18:
IN BACKCHECK

[OH4A6A1-100 - RUDDER TRIM MECHANISM v5.pdf](https://github.com/jrsteensen/OpenHornet/files/11264394/OH4A6A1-100.-.RUDDER.TRIM.MECHANISM.v5.pdf)

jrsteensen, 2023-04-21:
Show on separate sheets so you can give them dedicated Parts lists
Ensure dash numbers are the dash number of the part.
[image]
[image]

Lazarus94, 2023-04-22:
IN BACKCHECK

[OH4A6A1-100 - RUDDER TRIM MECHANISM v6.pdf](https://github.com/jrsteensen/OpenHornet/files/11300502/OH4A6A1-100.-.RUDDER.TRIM.MECHANISM.v6.pdf)

jrsteensen, 2023-04-22:
Made required changes, and added to RC Release Package

[OH4A6A1-100 - RUDDER TRIM MECHANISM_-_v8.pdf](https://github.com/jrsteensen/OpenHornet/files/11302018/OH4A6A1-100.-.RUDDER.TRIM.MECHANISM_-_v8.pdf)

closed 2023-04-22 by jrsteensen