jrsteensen/OpenHornet #689: Cosmetic: OH1A8A4-1 - ASSY, HUD BIT BOX No-Go lens cap is wrong color

DeadMeatDCS, 2023-04-14:
### Summary:
Minor detail: NO-GO lens cap on HUD BIT box is currently red. It should probably be amber or yellow instead.
The indicator should light up amber according to NATOPS (I'm told) but the lens cap itself looks quite yellow according to photo reference.

### Expected Results/How It Should Work:
An alternative is the **CMC 441 ATP** amber lens cap. It looks quite orange when it is off however.

Were talking about a cosmetic item so the yellow **CMC 441 YTP** may actually be a better choice. I'd personally go for this one.
Off, it looks closer to photo reference but on it lights up quite white (def not amber). 

I think the off state is what matters most since the light will never turn on in game anyways unless you rig up it manually as a gimmick (not implemented in DCS, most likely never will be).

### Actual Results/How It Does Work:
**CMC 441 RTP** is the red lens cap erroneously featured in the current design.

### Screenshots:
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [X] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH1A8A4-1 - ASSY, HUD BIT BOX

jrsteensen, 2023-04-14:
This is a fantastic catch. I've verified it against multiple sources. Thanks @DeadMeatDCS. (Of course my red one just arrived from digikey today. Hah.)

jrsteensen, 2023-04-14:
Completed and closed in latest commit.

closed 2023-04-14 by jrsteensen