jrsteensen/OpenHornet #693: APU Ready Lens: End of Life (131A-303G)

Higgins-OH, 2023-04-15:
### Summary:
131A-303G is no longer produced, drop-in part needed.

### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [x] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [x] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH4A5A2-1 ASSY, APU ENG CRANK PANEL

jrsteensen, 2023-04-17:
Not obsolete yet, will cross bridge when it is.

Arribe, 2023-11-26:
Update: No longer in stock at Digikey https://www.digikey.com/en/products/detail/visual-communications-company-vcc/131a-303g/3150656

Granted Newark as 445, and Mouser has 195 left.

kbastronomics, 2023-11-26:
I wonder is the substitute digikey recommends is viable
https://www.digikey.com/en/products/detail/dialight/1280932003/2428769

Arribe, 2023-11-27:
We'll find out, I have it on order.

Arribe, 2023-12-01:
Answer is not without expanding the legend plate hole to fit the larger diameter. 
[image]
[image]