jrsteensen/OpenHornet #701: OpenHornet Improved Display Initiative

jrsteensen, 2023-04-18:
This issue is the define the requirements of what an "perfect" display for each of the displays required by OH looks like.

## Requirements
### General

1. Finish: Matte
2. View Angle: Wide (120°+)
3. Simple connections/wiring. (HDMI, 2pin power preferred.)

### DDI

1. Visible Area:
2. Max Physical Footprint:
3. Resolution:

### AMPCD

1. Visible Area:
2. Max Physical Footprint:
3. Resolution:

### IFEI Option A
#### Single Display

1. Visible Area:
2. Max Physical Footprint:
3. Resolution:

### IFEI Option B
#### Left Display
1. Visible Area:
2. Max Physical Footprint:
3. Resolution:

#### Right Display
1. Visible Area:
2. Max Physical Footprint:
3. Resolution:

kbastronomics, 2026-09-25:
here's information on 7.6" square IPS displays for the DDI/AMPCD
Regarding the 7.6-inch high brightness display :
TTG076XST-01 (no touch):
Sample order of 1-10pcs, EXW unit price is US$50
If need the capacitive touch panel (CTP) by optical boding, it requires an additional cost of US$15/pcs. Attached is the outline drawing TTG076XST-01C for details.
The HDMI adapter board TBG076HD-01, supports HDMI for display, USB for power and touch:
Sample order of 1-10pcs, EXW unit price is US$20
Attached are datasheets for detail
Hence,
3sets of  TTG076XST-01(no touch) + HDMI board, total EXW price is US$ 210;

[TTG076XST-01C_ Outline of 7.6-inch Square TFT LCD, 800x800, MIPI, with Capacitive Touch.pdf](https://github.com/user-attachments/files/32635482/TTG076XST-01C_.Outline.of.7.6-inch.Square.TFT.LCD.800x800.MIPI.with.Capacitive.Touch.pdf)
[TBG076HD-01_HDMI to MIPI 30 board, support Touch by USB.pdf](https://github.com/user-attachments/files/32635481/TBG076HD-01_HDMI.to.MIPI.30.board.support.Touch.by.USB.pdf)
[TTG076XST-01_7.6-inch Square TFT LCD, 800x800, MIPI, High Brightness of 1000nits(1).pdf](https://github.com/user-attachments/files/32635483/TTG076XST-01_7.6-inch.Square.TFT.LCD.800x800.MIPI.High.Brightness.of.1000nits.1.pdf)

kbastronomics, 2026-09-25:
[image]
[image]
[image]

all photo's courteous of ben

kbastronomics, 2026-09-25:
https://tailorpixels.com
Phone/WhatsApp: +86 18923451443