jrsteensen/OpenHornet #702: UFC 5.6 Update

jrsteensen, 2023-04-20:
## Requirements

- [ ] Update BL Circuit so all LEDs light evenly, and are controlled by a single PWM pin from the microcontroller.

kbastronomics, 2023-04-26:
on this item
"On spare microcontroller IO, add a two pin header with jumper for software config (DCS-BIOS vs HID/Joystick inputs)"
instead of using an I/O port for this why not PRESS and HOLD I/P BUTTON at POWER UP to select DCS-BIOS, PRESS and HOLD ECM Button to select HID,  I can remember the selection until it's changed

SCUBA82, 2023-05-27:
Regarding the backlight observations: 

As already mentioned on discord all MOSEFTs (Q1,Q2,Q3) have their source and drain pins swapped. 

This is a screenshot of the original schematic 
[image]

and here how it should be ...

[image]

**BL LEDS general:** 
Each BL LED pair should be current limited with an resistor. Around  33Ohm / pair.

**LEDs D3 and D34:** 

The resistor value of D34 seems to be undersized. It should be around 110 Ohm.
D3 is connected in series with LED of the IP switch. Since the switche LEDs have a different forward voltag than the other BL LEDs D3 is brigther. 

Switch LED brightness:
I've tested generic Aliexpress switches and also noticed that their LEDs are much dimmer. Testing them with different voltages outsided of the UFC PCB showed that I can not make them much brighter. So I don't think there's a major issue with them (besides the current limiting resistors)

jrsteensen, 2023-07-02:
# v5.6 PCB
## Requirements
### UFC PCB

- [ ] Update BL Circuit so all LEDs light evenly, and are controlled by a single PWM pin from the microcontroller. (KB will remain a separate line.)
- [x] Reverse K/A pins of the SCR BL. (K/1 = SCR_BL; A/2 = +5V)
- [ ] Add Silkscreen for K/A markings.
- [ ] Add resistor to SCR BL net, 20ohm, .5 watt.
- [ ] On spare microcontroller IO, add a two pin header with jumper for software config (DCS-BIOS vs HID/Joystick inputs)
- [ ] Additional two pin headers for additional config?
- [ ] Change analog input to correct pot pin order. 
  - Flip pins 2&3 on J3,J4 and J8 that will move the signal to the center pin and 3.3V to the outer pin.
- [ ] Add 2-3 more TP to GND for convenience during debug.
- [ ] Isolate and switch between VSUB and EXT PWR automatically when EXT PWR is applied.
  [image]
- [ ] Add level switched input to spare ESP32 GPIO from EXT PWR (to notify ESP32 when EXT PWR) is applied.
  - Intent is to disable backlight (or limit to low brightness) and limit display brightness so UFC function is preserved when running on USB connection only.
- [ ] Add IRQ line for AP KB GPIO Expander (Replace one GND)
- [ ] Add diodes to key switches/switch matrix to prevent ghosting from multiple simultaneous keypresses. (If possible, not critical for Hornet use case.)

### AP KB PCB

- [ ] Add DNP 4k7 pullup resistors on SDA/SCL lines. 
  - Make DNP, and manually install for debug/testing purposes only.
- [ ] Add IRQ line for AP KB GPIO Expander (Replace one GND)

kbastronomics, 2023-07-02:
On the LED Backlights,  The values are calculated but based on what's seen the Switch LEDs are just not as bright as the smd leds so that resistor may need to be adjusted.

the **Keyboard/Function Select Daughterboard** is duplicated in the **UFC_Function Select PCB (Keyboard/KB)** item.   probably just remove it

I would also suggest redrawing the LED circuit so it more cleanly shows how the led are hooked up,   LEDS in parallel with each parallel branch having 2 LEDS with a series resistor.

kbastronomics, 2023-07-03:
Instead of a DCS-BIOS vs HID/Joystick config .   we have to use DCS-BIOS for Backlighting and the Displays so that's a given.  the question then is what about the buttons.   let just go HID as Joystick buttons, all the knobs will be axis and then not worry about configuring anything?

jrsteensen, 2023-07-03:
> Instead of a DCS-BIOS vs HID/Joystick config . we have to use DCS-BIOS for Backlighting and the Displays so that's a given. the question then is what about the buttons. let just go HID as Joystick buttons, all the knobs will be axis and then not worry about configuring anything?

I'd say HID Joystick for all inputs, DCS-BIOS for all outputs from the PC to the UFC.

Is there any actual need to split the VBUS vs VCC? Or is that already handled on the S2-MINI or in the ESP32? Is there a simpler circuit? The one you took a picture of the schematic was for a high amperage application [source link](https://electronics.stackexchange.com/questions/274338/automatic-switching-between-external-5v-and-usb-bus-power-supplies)
  [image]

I was thinking of just doing something simple like this:
  [image]

This way we can sense when we have external power which is passed through a spare channel on the level splitter without any further components to a spare IO on the MC. Space is at an absolute premium to keep this a 2 layer PCB.

kbastronomics, 2023-07-03:
So looking at the S3 schematic,  VBUS goes thru a 3v3 regulator then direct to the VDD lines on the processor. then it also goes to a pin on the S2's header.   the +5V is VBUS   so if you supply +5v and tie to the VBUS you have VBUS being supplied by the PC and a external power source.   probably not a good idea. it won't damage the ESP32 but it could damage the PC.  The ESP32 itself does not use the +5v supply except to be regulated down to 3V3.

so we should protect VBUS and not allow the external supply feed VBUS

jrsteensen, 2023-07-03:
Okay, is there a simpler circuit? 

Or using the original one, smaller MOSFETs you would recommend for the BSS84/SP247N60S5? (Well stocked by JLC?)

kbastronomics, 2023-07-03:
depends on what the objective is.   
1.  don't' hook the EXT 5V to VBUS.  since we have to connect to DCS-BIOS the S2 will get its 3V3 thru VBUS which is what is output on pin32 of the S2.  it's not a input pin,  it's a output of VBUS
2. we need to allow either VBUS or 5V external to supply the 5V logic. in that case run 5v thru a Schottky and vbus thru one as well,  that will prevent ext from sending 5V to VBUS but allow either to suppo then combine the output to the +5V supply, 
3. need to select EXT 5v or VBUS to supply 5V rail.  but also allow 5V or VBUS to supply S2.  this is the more complicated circuit

jrsteensen, 2023-07-03:
@kbastronomics 
@SCUBA82 

Just posting this for a review:
[image]
Please pay close attention to the polarity of the FETs.

Intent is to switch between VBUS and VCC, send a 3.3V signal to the microcontroller to tell it that external power has been applied (to unlock full backlight/display brightness range), and have a external power status LED.

SCUBA82, 2023-07-03:
@jrsteensen  Sorry can't help with your review request but there are 2 other topics I'm thinking about: 

**GN1640T U5/U6 (COMM 1/2 Displays) connection**
For me it's not clear why the SEG pins on U5 and U6 are connected differently (other pins / reverse order).

[image]

I guess this is a legacy from prior UFC versions where only one GN1640T was used for both COMM displays?
Sure this can be handled in software. In my case I defined individual char arrays for both COMM displays. But I guess it would be more convenient to have them connected the same.

**Backlight LEDs **
I noticed you've specified an 33ohm resistor to each LED pair. The LEDs [datasheet ](https://datasheet.lcsc.com/szlcsc/Foshan-NationStar-Optoelectronics-FC-F1005UGK-520M5_C130723.pdf) states a typical forward voltage of 2.8V. Combining 2 LEDs together will exceed the 5V supply voltage. Not sure if it makes any difference.

SCUBA82, 2023-07-03:
Arghh... there are characters which can't be displayed correctly on the OP Displays since A&B and E&F are combined on a single output pin.
[image]

[image]

We still have one free GRID pin on the GN1640 but need a second one...

jrsteensen, 2023-07-03:
> Arghh... there are characters which can't be displayed correctly on the OP Displays since A&B and E&F are combined on a single output pin.
> We still have one free GRID pin on the GN1640 but need a second one...

Unfortunately this is a limitation using those ICs. I haven't found a more suitable IC, and that is the same IC that Tek is using. I made a quick bitmap to show the allowances for this:
If all segments are controlled independently:
[image]
With the tied AB/EF pins:
[image]

> @jrsteensen Sorry can't help with your review request but there are 2 other topics I'm thinking about:
>
> **GN1640T U5/U6 (COMM 1/2 Displays) connection** For me it's not clear why the SEG pins on U5 [...]

This was for routing expediency. But if it would simplify programming, I can look at reversing it?

> **Backlight LEDs ** I noticed you've specified an 33ohm resistor to each LED pair. The LEDs [datasheet [...]

The issue is when we get the key switch LEDs as part of the pair - might need to just add them when one or two of those LEDs are making up the pair.

kbastronomics, 2023-07-03:
@SCUBA82 

The Comm displays is a nuisance but not an issue in my mind.

On the C5/D5 Pins,
Yes I brought this up previously,   opened a issue on it,   I did propose that we could C5/D5 to I/O Pins and control manually.  the other way would be to use a TCA6424A or MAX6956  to replace the GN1640T but then we have to drive it completely ourselves 

As for the character set,  yeh the driver we have is from @Tekdept

That might be a custom build of the board as a test though,  not something for beta1

The backlight LEDs 33 Ohm seems really low to me,we need to limit current to 20mA and at 5.6V/.020 = 280ohm for 2 in series and 2.7/.020=135ohm for the single.  

if anyone wants to check my math feel free

SCUBA82, 2023-07-04:
@kbastronomics 
For my understanding the formular is : R = (Vs - Vf) / If. So (5 - (2 * 2.8)) / 0.02 = -30. So the next higher resistor value is 1Ohm.
That's why I mentioned the problematic combined forward voltage of the LEDs...

 So I guess there are 2 options. Increase source voltage for the backlights (like it's done on the DDIs). Or use one resistor for each led. 

Regarding the comm display: Since it's tested and working I guess it would be best to keep it as it is.

Regarding the OP Displays.: 
The last unused gn1640 pin could be used, in combination with an AND gate to switch B1 and E1. The only character which would be impossible to print would be "[".

I totally understand if you decide to have a feature freeze for 5.6 and having the alternative charset is realistic enough I think.

jrsteensen, 2023-07-04:
Honestly, @SCUBA82 - I want this iteration of the UFC to be right, and hopefully never touch it again. I'm rolling an experimental version converting to WS2812 backlighting to test, with standard high force tact switches. 

If you would like to submit a redline of what that circuit would look like to fully enable the displays, I'd appreciate it. Needs to be small though. Real estate is going to be at an absolute premium.

SCUBA82, 2023-07-04:
@jrsteensen  Referring to your character overview, segment B is only active when A is also active. The same is true for E and F. There is no combination where A + B are active while only F is active or vice versa. 
The only exception is character "[" where B and F are used explicit. So this is not a perfect solution... (Do you know in which menu it is displayed?)

So my idea is to define pin GRID16 as "B" and "E" in software. Then AND it together with GRID1 (A) and connect the output of the AND Gate to segment B of the display. This way segment B will only light up if GRID16 and GRID1 are active. 

Something like this: 
[image]

jrsteensen, 2023-07-04:
So, here is the experiment of the function select board, with WS2812B-2020 backlighting, and the high force omron tactiles. Will require a minor MCAD update for the button interface (switches are 3mm shorter than the previous ones. The goal is even, customizable backlighting to eliminate hotspots. So many LEDs are used not because we need the brightness, but because we need to be able to dim them to eliminate those hotspots. I'm sending this to @kbastronomics to order some test units, and if it works well, we will continue with updating the UFC with this.

[image]
[image]

The cap is only 100uF compared to the usual 1000uF, but I just couldn't shoehorn it on due to physical size constraints, and figure some smoothing is better than no power smoothing.

@SCUBA82 That makes a lot of sense. I'd like @kbastronomics opinion as well as far as the complexity it brings to code. Thanks for the support!

kbastronomics, 2023-07-04:
@SCUBA82 
I did think about doing something like that as well but wasn't sure the introduction of logic gates would be well received.
@jrsteensen This is a good solution with minimal changes to the existing library

on the UFC Function,   I take it the #LED is the Data IN for the 2312s?  should be a issue as long as they are the last one on the string (ie UFC board first then AP board) or you run a seperate IO pin for the AP board,  either works and I like it!
I wouldn't worry so much about the cap,   you could just put but the big cap on the UFC board,   you don't need one also on the AP board if running from same power supply.

kbastronomics, 2023-07-04:
@SCUBA82 & @jrsteensen,
I did think about one thing,  the grid pins are open drain,   you can't hook up a standard TTL AND gate you'll need a open drain output one,   something like a SN74AHC1G09

SCUBA82, 2023-07-13:
[UFC_TEST_ESP32S2.zip](https://github.com/jrsteensen/OpenHornet/files/12044225/UFC_TEST_ESP32S2.zip)

Here is my UFC v5.5 test sketch. It's by no means production ready but tests the following main features: 

[x] scratchpad -> DCS-BIOS integartion    
[x] OP Displays 1-5 -> DCS-BIOS integartion  
[x] COMM Displays 1 & 2 -> DCS-BIOS integartion 
[x] Rotary Encoder 1 & 2 -> Interrupt triggered; Serial print a counter 
[x] Keypad Matrix -> Interrupt triggered;  Serial print button name; TODO rework PRESS / RELEASE detection
[x] ADF Switch  -> Interrupt triggered;  Serial print switch status; TODO rework PRESS / RELEASE detection
[x] BIT Switch  -> Interrupt triggered;  Serial print switch status; TODO rework PRESS / RELEASE detection
[x] VOL1, VOL2, BRT potis -> Serial print values every 2 sec
[x] GO / NOGO LED -> blink GO LED 5 sec; blink NOGO LED 2 sec

jrsteensen, 2023-07-13:
> 
> Here is my UFC v5.5 test sketch. It's by no means production ready but tests the following main features: 
> 
>  (snip)
> 

This is awesome. Thank you!

jrsteensen, 2023-11-02:
@kbastronomics has several test articles ordered and plans a (rough) ETA of 1DEC23 to have a test article built out. (For what is now the v6 UFC PCB)

Between now and then I will update MCAD to account for the new OMRON switches and WS2812 backlighting.

jrsteensen, 2023-12-29:
Update UFC Main PCB with:

- [ ] 1. Add 1k resistor across pins 1 & 2 of the 5X OPT Display MOSFETS
- [ ] 2. Note to use tab of display to orient COMM display modules. (Tab up!)
- [ ] 3. Change scratchpad BL to 10ohm resistor and 0-1k ohm trimpot in series. (Mount trimpot on back for accessibility.)

jrsteensen, 2024-02-25:
Getting a new OPT Display design, which will break the cue digits out to its own digit/segment pins.

(This is tentative, and not for design reference as of now.)
[image]

As of now, I am moving the UFC to a future release. I don't want to wait another 6 months to launch the beta1 for OH and want to take the pressure off so we can get it right.

jrsteensen, 2024-04-08:
### UFC_Main v6.1.0

- [ ] Replace UFCOptionDigit with UFCOptionDigitV3
- [ ] Reroute DS2 thru DS6.
- [ ] Route Cue segment/digit to spare pins on Comm1/2 as appropriate (DS2 thru DS6)
- [ ] Note to use tab of display to orient COMM display modules on schematic notes. (Tab up!)
- [ ] Change scratchpad BL to 10ohm resistor and 0-1k ohm trimpot in series. (Mount trimpot on back for accessibility - ensure clearance in 3D model.
- [ ] Update version on PCB to v6.1.0.
- [ ] Generate new manufacturing files using JLCPCB plugin tools.
- [ ] Copy manufacturing files and schematic PDF to appropriate folder in `release` folder.
- [ ] Generate new STEP file (and ensure all 3D components are populated.)

jrsteensen, 2024-10-05:
## v6.1.1 Updates

- [ ] Increase OPT display pin holes .02mm (to give .04 total additional tolerance on vertical pin spacing.)
  - **Justification:** This batch of displays has .01-.03mm smaller vertical pin spacing between the two rows, which is still within tolerance, but our footprint is too tight. Can be installed as-is, but difficult.

kbastronomics, 2024-10-12:
Haven't tested yet but an observation with only USB Power connected
1.  5v, and 3.3V LED are on as expected
2. EXT Power LED is on (shouldn't be??)

SCUBA82, 2024-10-14:
### Description
As discussed on discord, the current "PWR SUPLLY SWITCH/BYPASS" design deviates from [this suggested circuite](https://electronics.stackexchange.com/questions/274338/automatic-switching-between-external-5v-and-usb-bus-power-supplies).
As a result the board was not powered correctly and issues were notified in the scratchpad area. 

### Schematics
**Here's a updated version of this circuite**
[image]

**PCB routing**

[image]

**Changes:** 
>  Schematics:
> - Q8
>    - changed Symbol to n-channel
>    - switched drain and source
>    - Updated Datasheet and description
> - Q4, Q6, Q7
>    - changed Part to SKML5203
>    - Updated Datasheet, description, LCSC and PN
>
> PCB:
> - Q8
>    - Updated routing according to schematic

### Simulation
A simulation of this circuite can be found [here](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWK0wHYwA4MDYCcAmBDfDXAZhOxABZIQkE6EBTAWjDACgB3cAkfPmrgw+AUKicASjQzjh7KoOF06telE0wEPAajHK+BiZF1lIww2TJKTZ7HMO06h0zLRUK4VFUVaaatC4-tpmFiBe+PoRYqa81p5iNnJepgAmEQ4xNGCJAuAAcgjo+JwATjF0oin4Vdiq4PBwnADmleD17R5aukJ0kdGpnADO3nl92Q0AZgCGADbDzCOy8iLGKuAgswtLAA45tsIIJKtkPVPtGP212dQQYNBnDyS1CNWv75wXZMd6Sb9Oe6PcDQF6QN5giHEL4CLB-WGOCTWEFPUHEcHvLgVZwCdQ4khyKpNUxtCj9GztH6xSQXKJJG5kykGejQLD4Km4O6lAAeAkgYlYwXZclYVGEYGEADUAEIAVQAymYbld2rgqHFVYkqicNWB+GqjKsNQT4Yy6ZJ4jdzVT4RrGeZhGa4XarXCzdEXS44fjnbp8Xi4Ea-blbiHXMHDgEXHZeHr6VV+OG2sdEUcThNVJxeawyJZqEh09Q5OLhABRAAaABUADrDAAKAHUpFn6II+Kh6G8+WpwMJq7sEJLa7o4yADYyDbr+O74zS1r3xaIx2KXCy2RyuSOlxLF+tJDIDeOxJPNOoyMCGqEKtUVthhEXNGAmiOTg+b5POPsbyGVTmkj1ryXQwOAMZ1wFqThcjEECERAO9YLENJmFmABXOYABdWDmZgMggS9YC4Np4JNB8TUzWNXzkb91VaOD72LBiegAe3AOR4LUSBcGCJAYCaXAfAQbBCE0aCUk4FiJTo09OLkHiUE2LggA)

Additionally I tested this circuite on a breadboard (with off the shelf components).
Testscenario: VBUS @ 3.3V (to make the volatge change obviouse); EXT PWR @5.12 V
Results: 
   - VBUS is distributed when no EXT PWR is supplied
   - Circuite switches to EXT PWR when supplied
   - EXT PWR LED turns on/off 
   - VCC_SNS switches 
 
**Messy Breadboard**
[image]
**VBUS @ 3.3V supplied only**
[image]
**VBUS @ 3.3V and EXT PWR @ 5.12 V supplied**
[image]
**Vaultage reading at VCC_SNS when EXT PWR supplied (0V when not)**
[image]

### Repository 
[Commits ](https://github.com/jrsteensen/OpenHornet/compare/UFC_v6.1.0...SCUBA82:OpenHornet:UFC_v6.1.0) are done in my fork. Pull Request may be done but changes are really minor.

closed 2024-10-19 by jrsteensen