jrsteensen/OpenHornet #704: Create Drawing OH5A5A1-100 - ASSY, DUAL CONCENTRIC SHAFT POTENTIOMETER

Lazarus94, 2023-04-22:
Create Drawing OH5A5A1-100 - ASSY, DUAL CONCENTRIC SHAFT POTENTIOMETER

Lazarus94, 2023-04-22:
Add to project

Lazarus94, 2023-04-22:
IN PROGRESS

Lazarus94, 2023-04-23:
IN CHECK

[OH5A5A1-100 - ASSY, DUAL CONCENTRIC SHAFT POTENTIOMETER v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11304036/OH5A5A1-100.-.ASSY.DUAL.CONCENTRIC.SHAFT.POTENTIOMETER.v2.pdf)

jrsteensen, 2023-04-23:
IN BACKCHECK

[OH5A5A1-100 - ASSY, DUAL CONCENTRIC SHAFT POTENTIOMETER_-_v2-CHK1.pdf](https://github.com/jrsteensen/OpenHornet/files/11304151/OH5A5A1-100.-.ASSY.DUAL.CONCENTRIC.SHAFT.POTENTIOMETER_-_v2-CHK1.pdf)

Lazarus94, 2023-04-25:
IN BACKCHECK

[OH5A5A1-100 - ASSY, DUAL CONCENTRIC SHAFT POTENTIOMETER v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11324026/OH5A5A1-100.-.ASSY.DUAL.CONCENTRIC.SHAFT.POTENTIOMETER.v4.pdf)

jrsteensen, 2023-04-27:
Added to RC Release package and closed.

closed 2023-04-27 by jrsteensen