jrsteensen/OpenHornet #705: Create Drawing: OH5A5A1-200 - ASSY, BLEED AIR KNOB MECHANISM

Lazarus94, 2023-04-22:
Create Drawing: OH5A5A1-200 - ASSY, BLEED AIR KNOB MECHANISM

Lazarus94, 2023-04-22:
Add to project

Lazarus94, 2023-04-22:
IN PROGRESS

Exprezz0, 2023-04-25:
Fixed and saved.  Should be ready for you unless I spot something else.

Lazarus94, 2023-05-03:
IN CHECK

[OH5A5A1-200 - ASSY, BLEED AIR KNOB MECHANISM v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11383597/OH5A5A1-200.-.ASSY.BLEED.AIR.KNOB.MECHANISM.v4.pdf)

jrsteensen, 2023-05-05:
Add the following to drawing:

- [ ] Add modification note to A12405RNZQ 4-position rotary switch to make it continuous rotation.

  1. Add a stripe between switch halves with a paint marker or torque stripe compound to establish a rotational index of the halves of the switch.
  2. Pry 4X tabs apart gently with a flat head screwdriver. (Gently locking the switch in a vice by its shaft may prove helpful.)
  3. There are two spring loaded ball bearings inside that may launch out when opened. Gently remove the cover now.
  4. As shown in the picture, there are two tabs that interface with each other to form a stop.
  5. Grind the tab off the shaft side with a small file.
  6. Carefully reassemble switch.

- [ ] Add image to drawing.

[image]

jrsteensen, 2023-05-05:
IN BACKCHECK

[image]

Delete duplicate PL from SH2
[image]

Lazarus94, 2023-05-05:
IN BACKCHECK

[OH5A5A1-200 - ASSY, BLEED AIR KNOB MECHANISM v5.pdf](https://github.com/jrsteensen/OpenHornet/files/11408381/OH5A5A1-200.-.ASSY.BLEED.AIR.KNOB.MECHANISM.v5.pdf)

jrsteensen, 2023-05-05:
Add "OH5A5A1-200" in the title block on SH8.
Add view title to modification picture of "FN 7 MODIFICATION"

Lazarus94, 2023-05-05:
Should I also add a section view of the fully assembled shaft or is it unnecessary?

jrsteensen, 2023-05-05:
It could be helpful, I don't see any reason not to.

Lazarus94, 2023-05-05:
Cool will do.

jrsteensen, 2023-05-05:
Added to RC Release Package and Closed

closed 2023-05-05 by jrsteensen